NicoHood/HID #398: KEY_H =11 ERROR

Baumart, 2023-01-02:
Even if it has been asked many times, I can't get it right, this error always comes up:
In file included from c:\users\DUDU\documents\arduino\libraries\hid-project\src\hid-apis\KeyboardAPI.h:29:0,
                 from c:\users\DUDU\documents\arduino\libraries\hid-project\src\hid-apis\defaultkeyboardapi.h:27,
                 from C:\Users\DUDU\Documents\Arduino\libraries\HID-Project\src/SingleReport/BootKeyboard.h:30,
                 from C:\Users\DUDU\Documents\Arduino\libraries\HID-Project\src/HID-Project.h:50,
                 from C:\Users\DUDU\Documents\Arduino\MacroKeyPad\MacroKeyPad.ino:4:
c:\users\DUDUl\documents\arduino\libraries\hid-project\src\keyboardlayouts\improvedkeylayouts.h:54:21: note: #pragma message: Using default ASCII layout for keyboard modules
     #pragma message "Using default ASCII layout for keyboard modules"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\users\DUDU\documents\arduino\libraries\hid-project\src\keyboardlayouts\improvedkeylayouts.h:73:25: error: expected identifier before '=' token
     KEY_H               = 11,
                         ^
c:\users\DUDU\documents\arduino\libraries\hid-project\src\keyboardlayouts\improvedkeylayouts.h:73:25: error: expected '}' before '=' token
c:\users\DUDU\documents\arduino\libraries\hid-project\src\keyboardlayouts\improvedkeylayouts.h:73:25: error: expected unqualified-id before '=' token
c:\users\DUDU\documents\arduino\libraries\hid-project\src\keyboardlayouts\improvedkeylayouts.h:521:1: error: expected declaration before '}' token
 };
 ^

exit status 1

I also changed the version but it didn't work.

NicoHood, 2023-03-12:
See https://github.com/NicoHood/HID/issues/390

arcDaniel84, 2023-11-19:
I get the same messages while compiling, with PlatformIO and Arduino IDE2. But it can be uploaded and it works. Very strange.