NicoHood/HID #400: Arduino IDE 2.0.4 - what does this mean

aadrians1, 2023-01-17:
In file included from c:\users\admin1\documents\arduino\libraries\hid-2.8.4\src\hid-apis\KeyboardAPI.h:29:0,
                 from c:\users\admin1\documents\arduino\libraries\hid-2.8.4\src\hid-apis\defaultkeyboardapi.h:27,
                 from C:\Users\admin1\Documents\Arduino\libraries\HID-2.8.4\src/SingleReport/BootKeyboard.h:30,
                 from C:\Users\admin1\Documents\Arduino\libraries\HID-2.8.4\src/HID-Project.h:50,
                 from C:\Users\admin1\AppData\Local\Temp\.arduinoIDE-unsaved2023017-8044-34nlyz.x4lmg\sketch_jan17a\sketch_jan17a.ino:2:
c:\users\admin1\documents\arduino\libraries\hid-2.8.4\src\keyboardlayouts\improvedkeylayouts.h:54:21: note: #pragma message: Using default ASCII layout for keyboard modules
     #pragma message "Using default ASCII layout for keyboard modules"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\admin1\AppData\Local\Temp\.arduinoIDE-unsaved2023017-8044-34nlyz.x4lmg\sketch_jan17a\sketch_jan17a.ino:1:0:
C:\Users\admin1\AppData\Local\Arduino15\libraries\Keyboard\src/Keyboard.h:54:27: error: expected identifier before numeric constant
 #define KEY_RETURN        0xB0
                           ^
c:\users\admin1\documents\arduino\libraries\hid-2.8.4\src\keyboardlayouts\improvedkeylayouts.h:103:5: note: in expansion of macro 'KEY_RETURN'
     KEY_RETURN          = 40, // Alias
     ^~~~~~~~~~
C:\Users\admin1\AppData\Local\Arduino15\libraries\Keyboard\src/Keyboard.h:54:27: error: expected '}' before numeric constant
 #define KEY_RETURN        0xB0
                           ^
c:\users\admin1\documents\arduino\libraries\hid-2.8.4\src\keyboardlayouts\improvedkeylayouts.h:103:5: note: in expansion of macro 'KEY_RETURN'
     KEY_RETURN          = 40, // Alias
     ^~~~~~~~~~
C:\Users\admin1\AppData\Local\Arduino15\libraries\Keyboard\src/Keyboard.h:54:27: error: expected unqualified-id before numeric constant
 #define KEY_RETURN        0xB0
                           ^
c:\users\admin1\documents\arduino\libraries\hid-2.8.4\src\keyboardlayouts\improvedkeylayouts.h:103:5: note: in expansion of macro 'KEY_RETURN'
     KEY_RETURN          = 40, // Alias
     ^~~~~~~~~~
In file included from c:\users\admin1\documents\arduino\libraries\hid-2.8.4\src\hid-apis\KeyboardAPI.h:29:0,
                 from c:\users\admin1\documents\arduino\libraries\hid-2.8.4\src\hid-apis\defaultkeyboardapi.h:27,
                 from C:\Users\admin1\Documents\Arduino\libraries\HID-2.8.4\src/SingleReport/BootKeyboard.h:30,
                 from C:\Users\admin1\Documents\Arduino\libraries\HID-2.8.4\src/HID-Project.h:50,
                 from C:\Users\admin1\AppData\Local\Temp\.arduinoIDE-unsaved2023017-8044-34nlyz.x4lmg\sketch_jan17a\sketch_jan17a.ino:2:
c:\users\admin1\documents\arduino\libraries\hid-2.8.4\src\keyboardlayouts\improvedkeylayouts.h:521:1: error: expected declaration before '}' token
 };

nexposito, 2023-03-03:
I got the same error, could you figure it out what is the problem?

NicoHood, 2023-03-12:
See https://github.com/NicoHood/HID/issues/390

Eliaszszsz, 2024-02-22:
FOUND THE FIX!!!!!
its soooo easy. all you have to do is use the HID-project library at the 2.8.4 version. Hope i helped1

NamanS999, 2025-12-10:
I'm trying to use the library in arduino ide 2.3.6 and in arduino ide 1.8.19, and there is an error in both versions, how can I fix it?