NicoHood/HID #404: System.write(SYSTEM_WAKE_UP); applies to every button press

imstupidwithlinux, 2023-02-07:
i have 6 buttons connected to my pro micro and on button 6 i have places this to be able to turn on my pc.

System.write(SYSTEM_WAKE_UP);

after a few days of using it like that i noticed that im able to press any of the 6 buttons and my pc will boot when fully shut down. is this a me problem or a bios settings problem or a HID project problem

[code.txt](https://github.com/NicoHood/HID/files/10677123/code.txt)
[image]

imstupidwithlinux, 2023-02-08:
Update: I can do this with every usb device, mouse, keyboard. So how is it supposed to work then? I read that the pcie option too supports usb but it just won't do anything if usb is disabled.

NicoHood, 2023-02-09:
I dont understand your question. that call that you described will wakeup the pc.

imstupidwithlinux, 2023-02-09:
wake up from what state? S3? S5? im trying to use this instead of pressing the power button on my case

NicoHood, 2023-02-10:
You can only wakeup from sleep state, not from a real shutdown. If i remember correct, sleep was S3 and shutdown S5.