NicoHood/HID #407: multi-media keys on windows

WoutVV, 2023-02-27:
I wrote a code for my arduino leonardo. everything seems to be working fine except for 'HID_KEYBOARD_VOLUME_UP' , 'HID_KEYBOARD_VOLUME_DOWN' and 'HID_KEYBOARD_MUTE'. I remember reading that multi-media keys don't work on windows and only does on linux. I can't remember where though, sorry. It would be nice if anyone could help

AdverD, 2023-03-05:
Multimedia keys do work on Windows, but code examples and wiki can be kinda misleading.
The ImprovedKeyboard example contains a comment 
 `// Only works with the lower 255 keys and on linux only.`
But it's true only for Keyboard class. 

There are Consumer  and SingleConsumer classes that work for media keys on Windows pretty well.  
Use ` Consumer.write(MEDIA_VOLUME_UP); `  or whatever media key you need.
The list of key aliases can be found in [ConsumerAPI.h](https://github.com/NicoHood/HID/blob/master/src/HID-APIs/ConsumerAPI.h)

In regards to Consumer or SingleConsumer there is a bried explanation of USB endpoints in [API-Documentation](https://github.com/NicoHood/HID/wiki/API-Documentation)

WoutVV, 2023-03-06:
> Multimedia keys do work on Windows, but code examples and wiki can be kinda misleading. The ImprovedKeyboard example contains a comment `// Only works with the lower 255 keys and on linux only.` But it's true only for Keyboard class.
> 
> There are Consumer and SingleConsumer classes that work for media keys on Windows pretty well. Use `Consumer.write(MEDIA_VOLUME_UP);` or whatever media key you need. The list of key aliases can be found in [ConsumerAPI.h](https://github.com/NicoHood/HID/blob/master/src/HID-APIs/ConsumerAPI.h)
> 
> In regards to Consumer or SingleConsumer there is a brief explanation of USB endpoints in [API-Documentation](https://github.com/NicoHood/HID/wiki/API-Documentation)

TYSM this was the last part of my code that didn't work and now it does.

closed 2023-03-06 by WoutVV