NicoHood/HID #411: Fails to build/link using Arduino SAMD Boards version 1.8.13

aerobaticant, 2023-03-31:
As the title says, builds fail with **Arduino SAMD Boards (32-bits ARM Cortex M0+)** version **1.8.13**.
Reverting to version **1.8.9** builds works correctly.

I am using **HID-Project** version **2.8.4**

I believe this could be the same as issue #335 but hopefully a clearer title and more succinct description.

Errors:
```
C:\Users\Ant\Documents\Arduino\My-Project\build\libraries\HID-Project\HID-Project.a(BootKeyboard.cpp.o): In function `BootKeyboard_::getInterface(unsigned char*)':
C:\Users\Ant\Documents\Arduino\libraries\HID-Project\src\SingleReport/BootKeyboard.cpp:88: undefined reference to `USB_SendControl(unsigned char, void const*, int)'
C:\Users\Ant\Documents\Arduino\My-Project\build\libraries\HID-Project\HID-Project.a(BootKeyboard.cpp.o): In function `BootKeyboard_::send()':
C:\Users\Ant\Documents\Arduino\libraries\HID-Project\src\SingleReport/BootKeyboard.cpp:205: undefined reference to `USB_Send(unsigned char, void const*, int)'
C:\Users\Ant\Documents\Arduino\My-Project\build\libraries\HID-Project\HID-Project.a(BootKeyboard.cpp.o): In function `BootKeyboard_::getDescriptor(arduino::USBSetup&)':
C:\Users\Ant\Documents\Arduino\libraries\HID-Project\src\SingleReport/BootKeyboard.cpp:107: undefined reference to `USB_SendControl(unsigned char, void const*, int)'
C:\Users\Ant\Documents\Arduino\My-Project\build\libraries\HID-Project\HID-Project.a(BootKeyboard.cpp.o): In function `BootKeyboard_::setup(arduino::USBSetup&)':
C:\Users\Ant\Documents\Arduino\libraries\HID-Project\src\SingleReport/BootKeyboard.cpp:165: undefined reference to `USB_RecvControl(void*, int)'
C:\Users\Ant\Documents\Arduino\libraries\HID-Project\src\SingleReport/BootKeyboard.cpp:186: undefined reference to `USB_RecvControl(void*, int)'
collect2.exe: error: ld returned 1 exit status
```

mcuee, 2023-06-17:
For SAMD21, maybe you will have better luck with another library.
https://github.com/adafruit/Adafruit_TinyUSB_Arduino

I have no isssues running the example here from hidapitester.
https://github.com/todbot/hidapitester/tree/master/test_hardware/hidtest_tinyusb

Constannnnnt, 2023-11-17:
Reverting to version 1.8.9 builds aslo works for me.